wasilukm/hoymiles-mqtt #2: Jak wykonać instalację na home assistant ?

ZbyszekT1, 2022-06-23:
Jak wykonać instalację na Home Assistant ? Postawiony na wirtualnej maszynie bez linuxa. Czy jest możliwa instalacja z home assistanta lub jego konsoli. Proszę o podstawowe informację wykonania instalacji do powstania encji. Dalej sobie już poradzę. Pozdrawiam.

* Hoymiles MQTT version:
* Python version:
* Operating System:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

wasilukm, 2022-06-25:
Potrzebujesz:
- broker MQTT https://www.home-assistant.io/docs/mqtt/broker
- addon MQTT w Home Assistant https://www.home-assistant.io/integrations/mqtt/
- zainstalować Pythona https://www.python.org (jeśli nie ma)
- zainstalować hoymiles-mqtt:
  `python -m pip install hoymiles-mqtt`
- uruchomić hoymiles-mqtt z odpowiednimi parametrami, np.:
  `python -m hoymiles_mqtt --mqtt-broker <IP address brokera MQTT> --dtu-host <IP adres DTU>`

W przypadku Windowsa wydaje mi się, że ostatnią komendę można dodać do Task Schedulera żeby narzędzie samo się uruchamiało przy starcie systemu (https://www.windowscentral.com/how-create-automated-task-using-task-scheduler-windows-10)

kris-lwks, 2022-06-25:
U mnie działa świetnie z `docker compose up`. W docker-compose.yml mam:

```
version: "3.8"

services:

  # Home Assistant home automation service
  homeassistant:
    container_name: homeassistant
    image: "ghcr.io/home-assistant/home-assistant:stable"
    ports:
      - "8123:8123"
    volumes:
      - ./homeassistant:/config
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    privileged: true

  # MQQT broker
  mosquitto:
    image: eclipse-mosquitto
    container_name: mosquito
    restart: unless-stopped
    volumes:
      - ./mosquitto:/etc/mosquitto
    ports:
      - 1883:1883
    command: mosquitto -v -c /etc/mosquitto/mosquitto.conf

  # Publishes data from Hoymiles DTU to the MQTT broker
  hoymiles_mqtt:
    image: hoymiles_mqtt # first run: docker build https://github.com/wasilukm/hoymiles-mqtt.git\#v0.1.0 -t hoymiles_mqtt
    restart: unless-stopped
    environment:
      MQTT_BROKER: mosquitto
      DTU_HOST: 192.168.8.133
      QUERY_PERIOD: 15
    depends_on:
      - mosquitto
```

W tym samym katalogu miałem również

- pusty folder `homeassistant` który zostanie wypełniony plikami konfiguracyjnymi przy pierwszym uruchomieniu
- folder `mosquitto` z pojedynczym plikiem `mosquitto.conf`:

```
persistence true
persistence_location /mosquitto/data/
listener 1883
allow_anonymous true
```

-------------

Jedyna uwaga to kiedy wyłączam instalacje i mikroinwertery się odcinają to home assistant podaje ostatnią wartość dla wszystkich parametrów zamiast a hoymiles-mqtt raportuje błąd modbusa :

```
development-hoymiles_mqtt-1  | Exception in thread Thread-124:
development-hoymiles_mqtt-1  | Traceback (most recent call last):
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/threading.py", line 980, in _bootstrap_inner
development-hoymiles_mqtt-1  |     self.run()
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/threading.py", line 917, in run
development-hoymiles_mqtt-1  |     self._target(*self._args, **self._kwargs)
development-hoymiles_mqtt-1  |   File "/hoymiles-mqtt/hoymiles_mqtt/runners.py", line 40, in execute
development-hoymiles_mqtt-1  |     plant_data = self._modbus_client.plant_data
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 111, in plant_data
development-hoymiles_mqtt-1  |     microinverter_data = self.microinverter_data
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 88, in microinverter_data
development-hoymiles_mqtt-1  |     result = self._read_registers(client, start_address, 20, self._unit_id)
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 74, in _read_registers
development-hoymiles_mqtt-1  |     raise result
development-hoymiles_mqtt-1  | pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

[image]

wasilukm, 2022-06-26:
@kris-lwks dzięki za podzielenie się swoją konfiguracją. Zgadzam się, użycie dockera jest dobrym rozwiązaniem.

Co do drugiego wątku, stwórz proszę osobne zgłoszenie. Jeśli to możliwe najlepiej po angielsku.

bigdata00007, 2022-06-28:
Cześć,
A ja mam pytanie, czy Twoje rozwiązanie jest wstanie przechwytywać wszystkie alarmy z DTU - np. przekroczenie napięcia 253V?

Obecnie muszę zaorać swoją konfiguracje, ponieważ używam jakiegoś innego rozwiązania (innego skryptu w HA) bo niestety z instalacją Twojego sobie nie poradziłem :( - dopiero zaczynam przygodę z HA

kris-lwks, 2022-06-28:
@bigdata00007 hoymiles-mqtt to integracja z mikroinwerterami firmy Hoymiles na których napięcie wynosi max 60V.
Zakładam że masz standardowy falownik innej firmy?

bigdata00007, 2022-06-28:
Posiadam Hoymiles HMT-1800, chodzi o alarmy jakie pojawiają się, w sieci,
microinverter to wychwytuje (widać to tylko przez stronę www, a nie w
aplikacji, chyba że masz dostęp instalatora) i się wyłącza. Do tego
posiadam DTU PRO-S.

W załączeniu przekazuję dokumentację jaką dostałem od dystrybutora.

wasilukm, 2022-06-29:
Alarmy są dostępne dla każdego microinvertera z osobna.

Zamykam issue gdyż mocno zboczyło z tematu. W razie dalszych problemów proszę załozyć nowe.