wasilukm/hoymiles-mqtt #3: [Input/Output] Modbus Error: [Invalid Message] No response received

kris-lwks, 2022-06-27:
* Hoymiles MQTT version: v0.1.0
* Python version: 3.9.13
* Operating System: fedora coreos

Upon disconnecting the PV installation Hoymiles microinverters shut down and most of the MQQT parameters should change as the power production ceases.

What actually happens is that Home Assistant continues displaying the last known value (as on attached screenshot) and the modbus error appears in hoymiles-mqtt. I would expect that some of parameters, like temperature, are still valid and some should be either 0 or "no data" (like power production).

```
development-hoymiles_mqtt-1  | Exception in thread Thread-124:
development-hoymiles_mqtt-1  | Traceback (most recent call last):
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/threading.py", line 980, in _bootstrap_inner
development-hoymiles_mqtt-1  |     self.run()
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/threading.py", line 917, in run
development-hoymiles_mqtt-1  |     self._target(*self._args, **self._kwargs)
development-hoymiles_mqtt-1  |   File "/hoymiles-mqtt/hoymiles_mqtt/runners.py", line 40, in execute
development-hoymiles_mqtt-1  |     plant_data = self._modbus_client.plant_data
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 111, in plant_data
development-hoymiles_mqtt-1  |     microinverter_data = self.microinverter_data
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 88, in microinverter_data
development-hoymiles_mqtt-1  |     result = self._read_registers(client, start_address, 20, self._unit_id)
development-hoymiles_mqtt-1  |   File "/usr/local/lib/python3.9/site-packages/hoymiles_modbus/client.py", line 74, in _read_registers
development-hoymiles_mqtt-1  |     raise result
development-hoymiles_mqtt-1  | pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

Let me know if there is anything else I can try to test.
I am very new to modbus but it got my interest and I can dig into this problem some more as I intend to connect these metrics with power consumption metrics (for instance from Zamel MEW-01).

[image]

kris-lwks, 2022-06-28:
That should work I think. If we can't request data from the DTU it is safe to assume the data is unavailable.

closed 2022-06-29 by wasilukm

wasilukm, 2022-06-29:
In 0.2.0 I've added option expire-after. With it you can specify time when entities shall become unavailable in case of communication issues (no updates sent to Home Assistant). Here is an example chart for some entity with communication loss:
[image]

PR #4 (2022-06-29): Add support for expire_after
Add support for entities expiration, DTU or microinverter entities may become `unavailable` in Home Assistant after a certain number of seconds when there is no update from `hoymiles-mqtt`

Resolves #3