wasilukm/hoymiles-mqtt #6: 2 Inverter with 8 PV modules but only 2 found

FrodoVDR, 2022-10-08:
* Hoymiles MQTT version: 0.20
* Python version: 
* Operating System: Docker

### Description

Cool tool. If I see it correctly, only one connection is displayed per inverter.

I have two HMS-1800-t4 with 8 solar panels in operation. However, only port 4 is displayed. Is it possible to query the values of all 8 PV modules?

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

FrodoVDR, 2022-10-09:
After taking a closer look at the output via NodeRed, I saw that I get all the data.
Only Homessistant only shows port 4 because that is always the last module found.

The issue can be closed.

wasilukm, 2022-10-09:
@FrodoVDR thank you for the feedback. I have no experience with HM series so it would be helpful if you could provide row data from your installation with the following Python code:
```
from hoymiles_modbus.client import HoymilesModbusTCP
from pprint import pprint

plant_data = HoymilesModbusTCP('1.2.3.4').plant_data
pprint(plant_data)
```
You may need to install hoymiles-modbus library and replace IP address in the script.

I'm reopening the issue as it seems to be still valid.

FrodoVDR, 2022-10-13:
You have to describe this in more detail, I don't know python where to add the code. I have this running with Docker.

Regardless, my DTU-Pro-s has partially hung today, only after a restart by removing the power supply all values were delivered again.

wasilukm, 2022-10-16:
No problem. We can try reading the data from MQTT.
Execute command:

mosquitto_sub -h <ip_of_your_mqtt_broker> -v -t "homeassistant/hoymiles_mqtt/#" -u <mqtt_user> -P <mqtt_password>

Skip the two last parameters if your MQTT broker doesn't require authentication.

Wait until the data is published and send me the output.

If you don't have mosquitto_sub, install https://mosquitto.org/download/

closed 2022-10-30 by wasilukm