wasilukm/hoymiles-mqtt #9: No response received - ever

VincentSC, 2022-11-24:
* Hoymiles MQTT version: 0.3.0
* Python version: 3
* Operating System: Raspbian GNU/Linux 10 (buster)
* DTU: PRO-S
* DTU HW-version: H09.06.01
* DTU SW-version: V00.02.14

### Description

Just trying to connect for the first time. I get this error 100% of the time: `pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)`

The Hoymiles app works well, and can read the data from all.

This software needs `libseccomp2` to be installed, else it won't work. I fixed that already.

Restarted the device, but neither helped.

It seems to connect to the DTU correctly, as other ip-addresses give another error.

### What I Did

```
python3 -m hoymiles_mqtt --mqtt-broker localhost --dtu-host 192.168.50.99 --query-period 5
```
Also tried the HM microinverter-type. When I add `--comm-retry-on-empty True` it keeps on looping.

VincentSC, 2022-11-24:
Solution: upgrade to DTU SW-version to V00.02.15 (V0.2.0F).

Would be a good thing for the documentation.

wasilukm, 2022-11-26:
@VincentSC thank you for the feedback. I've added troubleshooting to readme based on your comments

closed 2022-11-26 by wasilukm

tudstudent, 2023-04-21:
@wasilukm and @VincentSC 
Just adding to this topic, recent experience. Even with latest firmware on the DTU it keeps locking up.
Exact same message as above.

python 3.10
DTU Hardware rev - H09.06.1
DTU Software - V00.02.15

Even though the DTU is nice for info, the connection drops seem to be quite often.
(Even without mod-bus)

Seems to be firmware, as the network of the DTU is still online -> e.g. ping  is working
3 Solutions I see as potential solutions:
- Power Cycle in the night (of the DTU)
- Mail  to Hoymiles with this unstable behavior
- Switch to an alternative data read solution (German solution) -> meaning getting rid of DTU for monitoring purposes)

Do you have ideas (ever since)

VincentSC, 2023-04-21:
Same experience right now: the DTU is super instable. Connections to the cloud drop a lot here - sometimes twice a day, sometimes it skips 1 or 2 days.

- The lock-ups occur at random moments, but it probably could be programmed - I have to look at statistics.
- Emailing is never a bad idea. I was thinking of emailing the webshop about the faulty product
- The German solution is not yet available for the HMS and HMT, only the older HM-series.

VincentSC, 2023-04-22:
Mmm, is that effort different from this team? https://github.com/lumapu/ahoy and https://discord.com/channels/984173303147155506/993500210271629393

I decided to just buy the official DTU after I saw how slow they were progressing, and it was impossible for me to join in due to the lack of a structured approach. Other projects were very quiet, and assumed that would remain. I seemingly bet on the wrong team, as the Ahoy team still has "coming soon" :) 

From what I see (and remember), building the hardware is relatively easy. So going to order parts this weekend if I did not do that already. Thanks for the tip!

Offtopic: if I'm reading `TU` and `Oke` correctly, I'm guessing you're also on Tweakers.net. Do you know of any electronics-makers-forum for such efforts like the Germans seem to have more often? https://www.circuitsonline.net/forum/search?query=hoymiles&mode=and&starter_user=&message_user=&topic_type=0&sort=0 shows less than Tweakers.

tudstudent, 2023-04-22:
Looking at the github sites it are different projects (probably share some code).

For me the openDTU looks a bit more professional.

Looking for gear now so I can start with that.
Mainly the antenna board sourcing seems to be challenging.

tudstudent, 2023-04-23:
@VincentSC I did send a mail to Hoymiles and they answered today. I need to send my ID so a technician will contact me and will investigate the matter. 
In the end the opendtu solution is nice, but the original DTU is easier (will try this route first).

VincentSC, 2023-05-02:
Did the same. Emailed them SW-version, HW-version and they asked for S/N. Now waiting for a magic solution.

tudstudent, 2023-05-03:
So far no result.
Almost have the feeling they have lockup in their TCP stack.
Once a day it resets and comes back online. No change in using fixed IP.

Would be interesting if the RS485 port is working fine, but as always not a lot of time to test.

tudstudent, 2023-05-05:
So far no result.
Almost have the feeling they have a lockup in their TCP stack.
Once a day it resets and comes back online. No change in using fixed IP.

Would be interesting if the RS485 port is working fine, but as always not a lot of time to test.

VincentSC, 2023-05-10:
Same problem here. They even stopped responding.

I ended up connecting a zigbee-powerplug, that turns off for 5 seconds and back on, every hour during the day.

If this works well for a week, I will let the restart trigger by a lack of mqtt-messages. Far from ideal, but I see no other option.

tudstudent, 2023-05-15:
For now disabled the modbus probing and it is more stable.
However no data in HA...
When more time I will most probably (if hoymiles will not come with a decent solution), switch to the opendtu.

PR #10 (2022-11-26): Add troubleshooting to README.md
Fixes #9