wasilukm/hoymiles-mqtt #14: Zeroed values every 15min

jtkacz, 2023-03-09:
* Hoymiles MQTT version: 0.4.0
* Python version: 3.10
* Operating System: Ubuntu 22.10 Linux 5.19.0-31-generic
* DTU-PRO-S V00.02.15 Hardware Ver.H09.06.01
* Hoymiles HMS-2000-4T 

### Description

Thank you for excellent piece of software!

Code works well but described elsewhere limitations in stability of Hoymiles DTU-PRO-S are significant. I observe every 4-5 days periodic drop-downs in readings during the day. 
I have attached the image of diagram presenting the issue. 

My startup parameters are:
python -m hoymiles_mqtt --microinverter-type HM --mqtt-broker 192.168.0.45 --mqtt-user xxx --mqtt-password xxx --dtu-host 192.168.0.157 --comm-close-comm-on-error True --query-period 100

To avoid total hang-ups I reset DTU every night at 2:13am. That solution resulted with stable work of DTU except the issue I describe here.

Observations:
- Drop-downs happens every 15min (obviously connected with 15min period of sending data to Hoymiles cloud) 
- Single drop-down can usually influence 1-4 of 5 inverters I have, not affected inverters have all values good
- Single drop-down zeroes all values for affected inverter (power, voltage, all ports,...)
- the issue is related to interval between queries set. Having default value 60s caused issue every day. 100s period resulted with issue every 4-5 days.

Knowing above I believe that it is possible to modify code of module in a way that faulty readings will be skipped - not send to mqqt. I strongly prefer to have no values at all than having them wrongly zeroed. 

I'm ready to modify code myself in quite primitive way checking the value of grid frequency against zero. Hopefully more smart solution exists!
[image]

wasilukm, 2023-03-12:
Hi, have you tried `--expire-after` option? Set the value to more than double the query period (for example 220 in your case)

jtkacz, 2023-03-13:
Hello. 
I did not. I believe that default is 'expire-after=infinite'. This is how I understand docs on this page: [https://www.home-assistant.io/integrations/sensor.mqtt/#expire_after](url)

Anyway I will try. I have just set this param. I will wait for failure and let you know.

Still the issue is not about missing data but wrong data received via modbus. At least once a week there is a day when almost every 15min zeroed values arrive from mqtt. This is single wrong sample every 15min. Next query results proper values. See attached data below.

I'm aware that that it will be difficult to differentiate in code logic between DTU failure and some regular situations.

[image]

wasilukm, 2023-03-13:
Please try the option anyway. What is the link status when it happens?

jtkacz, 2023-03-14:
Yes, I have configured 'expired' option yesterday evening.
Link status is still 1. 
But Operating status is 0. So it goes down from 3 to 0 when failure happens.

jtkacz, 2023-03-16:
Hi again.

I set "--expire-after" option to 10 minutes. But it did not help itself. Zeroed values still appeared in states and on graphs. 

Still you inspired me. I modified code (ha.py) in a way to not send selected sensitive states to mqtt when operating status of inverter is other than 3. On time of failure data are not send to mqtt. For 10minutes last values are used by HA. This is more than enough to cover the DTU failure.

Issue seems solved for me. :)

My DTU seems overloaded. It serves 5 inverters - 4 panels each, power meter via wire and modbus inquries via tcp. Unfortunately my DTU is 1Ghz version and I can't replace it with openDTU.

jtkacz, 2023-03-24:
I'm closing the issue.

wasilukm, 2023-03-26:
@jtkacz sorry for not responding for a long time. I can add an option to ignore values when the operating status differs from 3.

jtkacz, 2023-03-27:
Thank you for your attention and preparing special experimental version :)

But this is more complicated. Applying your patch results with numerous errors in log of HA when new code starts to ignore selected values due to `operating_status==3`:

> Template variable warning: 'dict object' has no attribute 'pv_voltage' when rendering '{{ value_json.pv_voltage }}'
> Template variable warning: 'dict object' has no attribute 'pv_current' when rendering '{{ value_json.pv_current }}'
> Template variable warning: 'dict object' has no attribute 'pv_power' when rendering '{{ value_json.pv_power }}'

That's because code uses common state_topic and common json payload for all values for each device_serial/port combination.  HA can't find ignored values and rises errors.

I have corrected that issue with change to sensor's config definition in ha.py: https://github.com/jtkacz/hoymiles-mqtt/commit/125d79fbd8726a0d63b8d21f5e9cd143399b883a

jtkacz, 2023-03-31:
There are some smaller issues still. Give me some time for testing. I will come back in a few days.

w-a-p, 2023-04-02:
Can I make a suggestion? Could you please at the end of the function "get_state" in ha.py check the value returned from the DTU for 'pv_power' and see whether it is the same as the sum of the 'pv_power' values from all of the microinverters?

In my experience the DTU continues collecting correct values for each microinverter, but "relatively often" (maybe once every 15 minutes as reported here) the value returned by the DTU for 'pv_power' is zero (or simply not consistent with what the microinverters report).

I know, ha.py is just trying to report the raw values so I could "fix" this in my mqtt application (in my case home assistant) - but it seems "cleaner" to do something in the code which tries to get these values from modbus rather than exposing to home assistant.

jtkacz, 2023-04-08:
@w-a-p  According to DTU modbus specification the only option to gather power value for whole plant is to read each microinverter and its port one by one, then sum everything to get total power. And that is what hoymiles_modbus\client.py  does to calculate PlantData.pv_power.

I fully share you opinion to correct DTU issue on lower level. Thus currently I fine-tune the code of hoymiles_mqtt package to hide zeroed values resulting from DTU under-performance. The main idea is that it is better to not send value at all than send wrongly zeroed value. 

I make experiments on fork https://github.com/jtkacz/hoymiles-mqtt/tree/ignore_values

jtkacz, 2023-04-16:
I haven't made any more updates to the code. Summarizing changes:
1. Skip sending ALL inverter port's measures to mqtt when operating_status <> 3. For temporary DTU issues it is better solution than setting states to unavailable, especially with parameter: --expire-after 600
    Benefit: avoid wrong data in database and ugly graphs with zeroed values;)
2. Temperature, grid voltage and frequency set to unavailable when operating_status <> 3
    Benefit: see p1
3. Send total values of energy production (from internal cache) to mqtt only if ALL inverters report operating_status == 3 
    Benefit: avoid issues in reporting of energy production by HA. State class set to total increasing hates incidental drop-downs.
4. Update module's internal energy cache when new value is larger than value currently stored in cache
    Benefit: see p3
5. Do not send to mqtt energy data on level of microinverter's ports (i.e. sensor.port_1_116481317589_today_production) when  energy is zero or operating_status == 3
    Benefit: see p3 

For me setup is working quite well now. 
@wasilukm Thank you again for this code. It is clean and well structured. It was pleasure to dig into it and update to my needs. If you find anything valuable in my changes please feel free to use ideas or code snippets.

wasilukm, 2023-04-16:
@jtkacz Thank you so much for your feedback, I will review your propositions and apply where feasible.

wasilukm, 2023-05-07:
@jtkacz why is important for you to prevent zero values, is operating_status<>3 not enough for today_production and total_production?
I also have doubts regarding p3, it would mean if one inverter is broken (which happened to me already) no data will be available for the whole installation. I think making sure that new data must be greater than cached is sufficient.

jtkacz, 2023-05-08:
> why is important for you to prevent zero values, is operating_status<>3 not enough for today_production and total_production?

I'm geek on diagrams ;) Look on diagram I attached on begin of this thread (power per panel). I hate this faulty drops to zero.

> I also have doubts regarding p3, it would mean if one inverter is broken (which happened to me already) no data will be available for the whole installation. I think making sure that new data must be greater than cached is sufficient.

I believe cache is not sufficient. Cache has limited persistence, restart of process clears it. I have 5 inverters, they wake up one after another  in the morning. If cache has been restarted at night then reported energy production increases in steps. Total increasing sensors interprets this steps wrongly as energy production. 

But I admit there is no one good way to solve this issues.

jtkacz, 2023-10-04:
Hi. I must admit that I lost patience to Hoymiles DTU after two devices were broken without any discoverable reason. And I have confirmed over other forums that this issue is quite frequent.
Since June I use OpenDTU. It works flawlessly for me after proper configuration on HA - no drops to zero, no hangs, no tricks.  

I propose to close this issue.

wasilukm, 2023-10-08:
Hi, I understand your frustration. Unfortunately, OpenDTU doesn't support my inverters, so I will have to stick to this solution,

Thank, you for all the feedback, recently I had no time to work on this project but I will keep this issue open at least for some time as I want to make some improvements based on this.

closed 2024-02-12 by wasilukm