wasilukm/hoymiles-mqtt #15: can't set MI_ENTITIES from environment variables

hymstra, 2023-06-27:
* Hoymiles MQTT version: 0.4.0
* Python version: 3.9
* Operating System: Ubuntu 22.04.1 LTS

### Description

I wanted to overload the MI_ENTITIES and PORT_ENTITIES by setting environment variables.
Expecting the environment variables to replace the default.
But it seems that the argument parser appends the environment variables to the default instead.

So from setting the environment variable:
MI_ENTITIES=['grid_voltage', 'temperature']

I end up with:
['grid_voltage', 'grid_frequency', 'temperature', 'operating_status', 'alarm_code', 'alarm_count', 'link_status', ["'grid_voltage'"]["'temperature'"]]

Is this a bug, or am I not doing this correctly?

wasilukm, 2023-07-09:
Hi, thank you for reporting this, yes this is a bug.

wasilukm, 2023-07-09:
With the fix, the following should be working fine:
_MI_ENTITIES='grid_voltage, temperature'_

PR #16 (2023-07-09): Fix parsing MI_ENTITIES and PORT_ENTITIES env variables.
Values specified in these variables were added to the default list of options instead of replacing them.

fixes #15
closed 2023-07-09 by wasilukm