wasilukm/hoymiles-mqtt #18: strings everywhere

foxos42, 2023-09-19:
* Hoymiles MQTT version: 0.5
* Python version: 3.9
* Operating System: alpine
* Docker: Yes

### Description

Hi,
nice work! I use v0.5 inside a docker container.

Would be great if data inside the json (int and float) are not quoted with "" so the data type is correct and not string.
For tools like node-red it will then be possible to convert the json correctly into a javascript object. And then I no longer have to cast everything manually. ;)

### What I Did

```
Just looked @ the data with mqtt explorer and saw the Issue ;)
```

wasilukm, 2023-10-08:
I'll check what can be done

wasilukm, 2023-10-22:
@foxos42 please try the latest main

foxos42, 2023-10-22:
Builded a hoymiles_mqtt:test

Works! Problem solved!
I don't see any other strings where they shouldn't be. Thank you!

before:
{
  "grid_voltage": "236.8",
  "grid_frequency": "49.98",
  "temperature": "13.9",
  "operating_status": "3",
  "alarm_code": "0",
  "alarm_count": "0",
  "link_status": "1"
}

after:
{
  "grid_voltage": 237,
  "grid_frequency": 49.99,
  "temperature": 13.9,
  "operating_status": 3,
  "alarm_code": 0,
  "alarm_count": 0,
  "link_status": 1
}

wasilukm, 2023-10-22:
Thank you for the quick feedback

closed 2023-10-22 by wasilukm