wasilukm/hoymiles-mqtt #36: Couldn't detect inverter type for serial 138291130512. Please report an issue.

polleke69, 2025-01-16:
* Hoymiles MQTT version: 0.9.0
* Python version: 3.12
* Operating System: Debian linux Bookworm

### Description
When running the latest version of homilies mqtt errors are produced.

At the moment I am running version 0.3.0 of homilies mqtt without any issues.

### What I Did

Ran homilies_mqtt:
python3.12 -m hoymiles_mqtt --dtu-host 192.168.178.157 --mqtt-broker 192.168.178.201 --query-period 30

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```
Failed to read data from DTU. Unknown failure type.
Traceback (most recent call last):
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/plum/data.py", line 209, in unpack
    item, offset = cls.__unpack__(buffer, 0, None)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 236, in __unpack__
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/hoymiles_modbus/datatypes.py", line 69, in _pv_current_type
    raise ValueError(f"Couldn't detect inverter type for serial {serial}. Please report an issue.")
ValueError: Couldn't detect inverter type for serial 138291130512. Please report an issue.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/plum/data.py", line 228, in unpack_and_dump
    item, offset = cls.__unpack__(buffer, 0, dmp.add_record(fmt=cls.name))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 274, in __unpack__
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/hoymiles_modbus/datatypes.py", line 69, in _pv_current_type
    raise ValueError(f"Couldn't detect inverter type for serial {serial}. Please report an issue.")
ValueError: Couldn't detect inverter type for serial 138291130512. Please report an issue.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/hoymiles_mqtt/runners.py", line 57, in execute
    plant_data = self._modbus_client.plant_data
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/hoymiles_modbus/client.py", line 90, in plant_data
    inverters = self.inverters
                ^^^^^^^^^^^^^^
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/hoymiles_modbus/client.py", line 68, in inverters
    inverter_data = InverterData.unpack(data_to_unpack)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/plum/data.py", line 216, in unpack
    cls.unpack_and_dump(buffer)
  File "/var/srv/homeassistant/hoymiles_mqtt_090/lib/python3.12/site-packages/plum/data.py", line 242, in unpack_and_dump
    raise exceptions.UnpackError(dmp, exc) from exc
plum.exceptions.UnpackError:

+--------+---------------+-----------------------------+-------------------+--------------------------+
| Offset | Access        | Value                       | Bytes             | Format                   |
+--------+---------------+-----------------------------+-------------------+--------------------------+
|        |               |                             |                   | InverterData (Structure) |
|  0     | data_type     | 12                          | 0c                | uint8                    |
|        | serial_number |                             |                   | serial_number_t          |
|  1     |   [0:6]       | b'\x13\x82\x91\x13\x05\x12' | 13 82 91 13 05 12 |                          |
|  7     | port_number   | 1                           | 01                | uint8                    |
|  8     | pv_voltage    | 32.9                        | 01 49             | udec16p1                 |
+--------+---------------+-----------------------------+-------------------+--------------------------+

ValueError occurred during unpack operation:

Couldn't detect inverter type for serial 138291130512. Please report an issue.
No DTU data received!

polleke69, 2025-01-16:
Hoymiles HMT-2250-6T

wasilukm, 2025-01-16:
Could you please install hoymiles_modbus library from the branch:
python -m pip install -U  git+https://github.com/wasilukm/hoymiles_modbus.git@missing_inverter_type

Run hoymiles_mqtt and check if the value of pv_current is correct.

polleke69, 2025-01-17:
That did the trick, running and publishing data as intended... Thank you!

wasilukm, 2025-01-19:
Thank you.
The fix is available in hoymiles_mqtt==0.10.0 and hoymiles_modbus==0.9.0

closed 2025-01-19 by wasilukm