wasilukm/hoymiles-mqtt #40: MQTT publishes sometimes fail with socket.gaierror: [Errno -3] Try again due to per-message DNS lookups

tudstudent, 2025-09-14:
**Version:** `hoymiles-mqtt v0.11.0` (Docker), Python 3.12, Paho MQTT (from image)
**Environment:** Docker on Linux, user-defined bridge network. Mosquitto broker in same host (published on `192.168.0.4:1883`). Other services healthy.

**Symptom:** Periodically a cycle logs:

```
Failed to publish data from DTU. Unknown failure type.
socket.gaierror: [Errno -3] Try again
DTU data received and published. 0 messages into mqtt://mosquitto:1883
```

Then the next cycles succeed (e.g., “51 messages into mqtt://mosquitto:1883”).

**What seems to be happening:**
The publisher uses `paho.mqtt.publish_single` for each message. That creates a **new connection per message**, which also triggers a **DNS lookup for the broker** each time. Under bursts (≈51 messages/cycle), Docker’s embedded DNS (127.0.0.11) occasionally returns `EAI_AGAIN` (“try again”), leading to a failed cycle with 0 messages.

**Why this matters:**
Data for that cycle is lost; logs fill with connect/disconnect noise. This is reproducible on Docker networks under moderate load.

**Workarounds tested:**

* Setting `MQTT_BROKER` to the broker’s **IP** (e.g., `192.168.0.4`) → eliminates DNS errors.
* Pointing the container’s DNS at the LAN resolver also helps, but still does many lookups.

**Proposed fix (any of these would solve it):**

1. Provide a **persistent MQTT client mode** (single connection reused per cycle) instead of per-message `publish_single`.
2. Or batch per cycle via **`paho.mqtt.publish.multiple([...])`** so there’s only **one** DNS lookup and connect per batch.
3. Optionally cache the resolved broker address or add retry/backoff for `EAI_AGAIN`.

**Expected behavior:**
Publishing a cycle worth of messages should not fail due to transient DNS resolver hiccups.

**Thanks!** Happy to test a branch with persistent client or batch publishing.

wasilukm, 2025-09-14:
Thank you for reporting this interesting case. I'll take a look at this.

wasilukm, 2025-09-16:
@tudstudent could you please try https://github.com/wasilukm/hoymiles-mqtt/pull/41

closed 2025-11-05 by wasilukm